pazi88/STM32_CAN #3: Extended identifier

ru3dnw, 2022-11-07:
Good afternoon! Please tell me how, using your library, to transmit /receive CAN messages with an extended identifier? What settings do I need to work with the extended identifier?
Thank you very much in advance!

pazi88, 2022-11-28:
Hi!

I think it should work as is. There is Boolean showing if id is extended. Though I haven't tested that, because I haven't needed that on my projects. I need to test/verify that when I have time.

ru3dnw, 2022-12-22:
I conducted a simple experiment - I took your "Write" example, changed their ID to 0x1FFFFFFF (29 bits). An oscilloscope with CAN decoding reads 0x7FF (11 bits). Is it possible to forcibly assign "msg.flags.extended ==1"? And how to do it?

pazi88, 2022-12-22:
> I conducted a simple experiment - I took your "Write" example, changed their ID to 0x1FFFFFFF (29 bits). An oscilloscope with CAN decoding reads 0x7FF (11 bits). Is it possible to forcibly assign "msg.flags.extended ==1"? And how to do it?

Oh yes you are right. You need to add "CAN_tx_msg.flags.extended = 1" to the struct in the write.ino, in addition to the longer ID to make it work as extended." By default it's 0 if not set separately.

Anyways I'll need to check the extended identifiers myself too to make sure those work. Good reminder.

ru3dnw, 2022-12-22:
Thanks! So it worked. True, I had to write "CAN_TX_msg.flags.extended = 1" in the sketch, although in the library the message flag is called as you suggested - "CAN_tx_msg.flags.extended = 1". Please add examples with the reception and transmission of "extended ID"

pazi88, 2022-12-22:
> Thanks! So it worked. True, I had to write "CAN_TX_msg.flags.extended = 1" in the sketch, although in the library the message flag is called as you suggested - "CAN_tx_msg.flags.extended = 1". Please add examples with the reception and transmission of "extended ID"

Yes, I tested that too and it works. I'll update the examples.

closed 2022-12-22 by pazi88