pazi88/STM32_CAN #12: Error compiling code

sam-tj, 2023-06-12:
Hi, there is an error while compiling sample write code .

```
STM32_CAN/STM32_CAN.h:123:26: error: expected ')' before '*' token
  123 |     STM32_CAN(CAN_TypeDef* canPort, CAN_PINS pins, RXQUEUE_TABLE rxSize = RX_SIZE_16, TXQUEUE_TABLE txSize = TX_SIZE_16);
```

sam-tj, 2023-06-18:
I am using arduino 2.0.4 with stm32 framework 2.5.0.

The code is one of the examples, CAN write.

My bosrd config is the following, 
- Nucleo 144
- Nucleo H723ZG

pazi88, 2023-06-19:
> 

> I am using arduino 2.0.4 with stm32 framework 2.5.0.
> 
> The code is one of the examples, CAN write.
> 
> My board config is the following,
> 
> * Nucleo 144
> * Nucleo H723ZG

I can reproduce this. Seem to be something to do with the board itself. Looking at the spec sheet, the H723 has FDCAN interface, instead of the plain old regular CAN. It probably needs some special handling (or worst case, totally different library). I'll look into that.

sam-tj, 2023-06-19:
Thanks, 
I will wait for your reply.

pazi88, 2023-06-19:
> Thanks, I will wait for your reply.

The HAL library file seems identical to others. This is probably some missing define somewhere. Difficulty is to try to find that...

sam-tj, 2023-06-19:
In any way I could help?

pazi88, 2023-06-19:
> In any way I could help?

Oh sorry I looked at wrong place. The H7xx indeed has FDCAN driver, not CAN driver. And those are different. I'll try to implement usage of that too in this library. But it looks like it requires quite a bit of work, so it won't be ready anytime soon.

benjineeering, 2024-03-06:
Same issue here with a STM32G431VBT6 would be nice to have this working on CANFD controller

Hitomamacs, 2024-03-18:
do we have an eta of when this feature will be released?

closed 2024-04-11 by pazi88