pazi88/STM32_CAN #13: PlatformIO Linker error when USBCON/CDC Enabled

gateseee, 2023-06-18:
Hello. I struggled a long time now with the STM32 to get it running via standard USB connection without using the STlink. 
For doing this, i had to add some flags in platformio.ini

```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_protocol = dfu
build_flags = 
    -D HAL_CAN_MODULE_ENABLED
    -D USBCON
	-D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
	-D HAL_PCD_MODULE_ENABLED
	-D USBD_VID=0x1EAF
	-D USBD_PID=0x0004
	-D USB_PRODUCT=\"bluepill\"
    
lib_deps = pazi88/STM32_CAN@^1.1.0
```

If i'm compiling now, as excample this minimal version: 

```
#include "STM32_CAN.h"
STM32_CAN Can( CAN1, ALT );
void setup() {
  return; 
}
void loop() {
  return; 
}
```

It throw's the following errors:

> Processing bluepill_f103c8 (platform: ststm32; board: bluepill_f103c8; framework: arduino)
> -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
> PLATFORM: ST STM32 (15.6.0) > BluePill F103C8
> HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
> DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
> PACKAGES:
>  - framework-arduinoststm32 @ 4.20400.0 (2.4.0)
>  - framework-cmsis @ 2.50700.210515 (5.7.0)
>  - tool-dfuutil @ 1.11.0
>  - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 13 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- STM32_CAN @ 1.1.0
> Building in release mode
> Compiling .pio\build\bluepill_f103c8\src\main.cpp.o
> Linking .pio\build\bluepill_f103c8\firmware.elf
> c:/users/thed0/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/bluepill_f103c8/libFrameworkArduino.a(usbd_conf.c.o): in function `USB_LP_CAN1_RX0_IRQHandler':
> usbd_conf.c:(.text.USB_LP_CAN1_RX0_IRQHandler+0x0): multiple definition of `USB_LP_CAN1_RX0_IRQHandler'; .pio/build/bluepill_f103c8/lib7ea/libSTM32_CAN.a(STM32_CAN.cpp.o):STM32_CAN.cpp:(.text.USB_LP_CAN1_RX0_IRQHandler+0x0): first defined here
> c:/users/thed0/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/bluepill_f103c8/libFrameworkArduino.a(usbd_conf.c.o): in function `USB_HP_CAN1_TX_IRQHandler':
> usbd_conf.c:(.text.USB_HP_CAN1_TX_IRQHandler+0x0): multiple definition of `USB_HP_CAN1_TX_IRQHandler'; .pio/build/bluepill_f103c8/lib7ea/libSTM32_CAN.a(STM32_CAN.cpp.o):STM32_CAN.cpp:(.text.USB_HP_CAN1_TX_IRQHandler+0x0): first defined here
> collect2.exe: error: ld returned 1 exit status
> *** [.pio\build\bluepill_f103c8\firmware.elf] Error 1

As soon as i remove the USB build flags, its compiling. 

Does anyone have an Idea? 

Chris

pazi88, 2023-06-18:
On F103 the usb and CAN share registers. So you can't use those at the same time. It's hardware limitation, so there is nothing you can do for it.

gateseee, 2023-06-18:
Is there any documentation about that, that i can dig further?

pazi88, 2023-06-18:
> Is there any documentation about that, that i can dig further?

There is something if you google the issue. But also if you carefully read the F103 datasheet from ST, there is indeed some registers that overlap between USB and CAN. Don't remember which ones, because it's so long time since I checked.

pazi88, 2023-06-19:
Great to know. I'll close this issue, since there isn't anything we can do for the hardware limitations.

closed 2023-06-19 by pazi88