pazi88/STM32_CAN #14: Error compiling  - STM32F042F6P6TR

pippopippo69, 2023-06-19:
Hi, i get this error, please help:

/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp: In member function 'void STM32_CAN::begin(bool)':
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:82:35: error: 'GPIO_AF9_CAN1' was not declared in this scope; did you mean 'GPIO_AF4_CAN'?
   82 |       GPIO_InitStruct.Alternate = GPIO_AF9_CAN1;
      |                                   ^~~~~~~~~~~~~
      |                                   GPIO_AF4_CAN
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:85:31: error: 'GPIO_SPEED_FREQ_VERY_HIGH' was not declared in this scope; did you mean 'GPIO_SPEED_FREQ_HIGH'?
   85 |       GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                               GPIO_SPEED_FREQ_HIGH
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:103:35: error: 'GPIO_AF9_CAN1' was not declared in this scope; did you mean 'GPIO_AF4_CAN'?
  103 |       GPIO_InitStruct.Alternate = GPIO_AF9_CAN1;
      |                                   ^~~~~~~~~~~~~
      |                                   GPIO_AF4_CAN
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:105:31: error: 'GPIO_SPEED_FREQ_VERY_HIGH' was not declared in this scope; did you mean 'GPIO_SPEED_FREQ_HIGH'?
  105 |       GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                               GPIO_SPEED_FREQ_HIGH
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:113:7: error: '__HAL_RCC_GPIOD_CLK_ENABLE' was not declared in this scope; did you mean 'HAL_RCC_MODULE_ENABLED'?
  113 |       __HAL_RCC_GPIOD_CLK_ENABLE();
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |       HAL_RCC_MODULE_ENABLED
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:124:35: error: 'GPIO_AF9_CAN1' was not declared in this scope; did you mean 'GPIO_AF4_CAN'?
  124 |       GPIO_InitStruct.Alternate = GPIO_AF9_CAN1;
      |                                   ^~~~~~~~~~~~~
      |                                   GPIO_AF4_CAN
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:126:31: error: 'GPIO_SPEED_FREQ_VERY_HIGH' was not declared in this scope; did you mean 'GPIO_SPEED_FREQ_HIGH'?
  126 |       GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                               GPIO_SPEED_FREQ_HIGH
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:129:21: error: 'GPIOD' was not declared in this scope; did you mean 'GPIOA'?
  129 |       HAL_GPIO_Init(GPIOD, &GPIO_InitStruct);
      |                     ^~~~~
      |                     GPIOA
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:133:26: error: 'CAN1_RX0_IRQn' was not declared in this scope
  133 |     HAL_NVIC_SetPriority(CAN1_RX0_IRQn, 15, 0); // 15 is lowest possible priority
      |                          ^~~~~~~~~~~~~
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:136:26: error: 'CAN1_TX_IRQn' was not declared in this scope
  136 |     HAL_NVIC_SetPriority(CAN1_TX_IRQn, 15, 0); // 15 is lowest possible priority
      |                          ^~~~~~~~~~~~
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp: In member function 'void STM32_CAN::enableMBInterrupts()':
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:739:24: error: 'CAN1_TX_IRQn' was not declared in this scope
  739 |     HAL_NVIC_EnableIRQ(CAN1_TX_IRQn);
      |                        ^~~~~~~~~~~~
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp: In member function 'void STM32_CAN::disableMBInterrupts()':
/Users/pippo/Documents/Arduino/libraries/STM32_CAN/STM32_CAN.cpp:759:25: error: 'CAN1_TX_IRQn' was not declared in this scope
  759 |     HAL_NVIC_DisableIRQ(CAN1_TX_IRQn);
      |                         ^~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

pazi88, 2023-06-21:
The errors looks quite simple. I'll look into this, if I can fix it.

pazi88, 2023-06-21:
Try now. It at least compiles. I don't have any F0-series boards to try on.

pippopippo69, 2023-06-26:
Hi, compiles and I can upload a sketch, need time to check functionality.

thanks a lot, I'll update here.

pippopippo69, 2023-06-27:
Hi, It does not work...

I tested with the digital oscilloscope and nothing cames out from STM32F042F6P6TR PA12 and PA11 pins.

I do not know if it can be usefull, the datasheet says: "Pin pair PA11/12 can be remapped in place of pin pair PA9/10 using the SYSCFG_CFGR1 register"...

pippopippo69, 2023-07-09:
Hi, still stucked on this problem..

Any hint?

Tks

EntropyK1d, 2023-07-27:
Hi, I'm compiling against the STM32F042K6 target, so should be the same. The library compiles and uploads. The "write" example sends a single CAN message, but then locks hard. The "read" example will loop until a CAN message arrives, and then lock hard.
Assuming that these examples work correctly on STM32F1x targets, my feeling is that there is something not quite right with the configuration for the STM32F0x target specifically on the RX side.
I will dig into this.

EntropyK1d, 2023-08-03:
Made enough progress for my requirements. Ran into a few issues. First was hardware - using a Nucleo32 F042K6 board the solder bridge to provide the 8MHz clock signal isn't in place.
Verified hardware set up using a simple CAN write from the StdPeriph_Examples compiled under IAR.
Noted that the settings for Time segments 1 and 2 are quite different compared to the StdPeriph_Examples.
Using the simple Write example, I was able to get a single CAN packet sent, with a 'true' return from Can.write, but the code would hard hang a few ms later.
If I commented out the line in Can.write to turn back on the CAN_IT_TX_MAILBOX_EMPTY interrupt, the code now sends out all CAN messages. Need to put in delays on calls to Can.write.
Good enough for my requirements. Hope this helps.

EntropyK1d, 2024-03-31:
Have got both Read and Write working on the Nucleo32 F042K6 board.
The issue is was to do with assigning and linking the interrupt routines. The STM32F0xx targets only have a single CAN interrupt and handler (CEC_CAN_IRQn and CEC_CAN_IRQHandler) used for both RX and TX. This needs to be defined against CAN1_IRQn at the top of STM32_CAN.h.
Then STM32_CAN.cpp needs to modified so that the interrupt functions and handlers are correctly assigned, and linked to the HAL routine (HAL_CAN_IRQHandler).
After doing this, it works great. Thank you!

pazi88, 2024-04-11:
This should be fixed

closed 2024-04-11 by pazi88