pazi88/STM32_CAN #15: Error compiling on F303

futuranet, 2023-08-04:
In Arduino IDE:
```
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp: In member function 'void STM32_CAN::begin(bool)':
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp:84:35: error: 'GPIO_AF9_CAN1' was not declared in this scope; did you mean 'GPIO_AF9_CAN'?
   84 |       GPIO_InitStruct.Alternate = GPIO_AF9_CAN1;
      |                                   ^~~~~~~~~~~~~
      |                                   GPIO_AF9_CAN
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp:112:35: error: 'GPIO_AF9_CAN1' was not declared in this scope; did you mean 'GPIO_AF9_CAN'?
  112 |       GPIO_InitStruct.Alternate = GPIO_AF9_CAN1;
      |                                   ^~~~~~~~~~~~~
      |                                   GPIO_AF9_CAN
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp:139:35: error: 'GPIO_AF9_CAN1' was not declared in this scope; did you mean 'GPIO_AF9_CAN'?
  139 |       GPIO_InitStruct.Alternate = GPIO_AF9_CAN1;
      |                                   ^~~~~~~~~~~~~
      |                                   GPIO_AF9_CAN
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp:141:31: error: 'GPIO_SPEED_FREQ_VERY_HIGH' was not declared in this scope; did you mean 'GPIO_SPEED_FREQ_HIGH'?
  141 |       GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                               GPIO_SPEED_FREQ_HIGH
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp:155:26: error: 'CAN1_RX0_IRQn' was not declared in this scope; did you mean 'CAN_RX0_IRQn'?
  155 |     HAL_NVIC_SetPriority(CAN1_RX0_IRQn, 15, 0); // 15 is lowest possible priority
      |                          ^~~~~~~~~~~~~
      |                          CAN_RX0_IRQn
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp:158:26: error: 'CAN1_TX_IRQn' was not declared in this scope; did you mean 'CAN_TX_IRQn'?
  158 |     HAL_NVIC_SetPriority(CAN1_TX_IRQn, 15, 0); // 15 is lowest possible priority
      |                          ^~~~~~~~~~~~
      |                          CAN_TX_IRQn
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp: In member function 'void STM32_CAN::enableMBInterrupts()':
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp:765:24: error: 'CAN1_TX_IRQn' was not declared in this scope; did you mean 'CAN_TX_IRQn'?
  765 |     HAL_NVIC_EnableIRQ(CAN1_TX_IRQn);
      |                        ^~~~~~~~~~~~
      |                        CAN_TX_IRQn
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp: In member function 'void STM32_CAN::disableMBInterrupts()':
C:\code\arduino\libraries\STM32_CAN\STM32_CAN.cpp:789:25: error: 'CAN1_TX_IRQn' was not declared in this scope; did you mean 'CAN_TX_IRQn'?
  789 |     HAL_NVIC_DisableIRQ(CAN1_TX_IRQn);
      |                         ^~~~~~~~~~~~
      |                         CAN_TX_IRQn
exit status 1
Error compiling for board Generic STM32F3 series.
```

futuranet, 2023-08-04:
Did a temporary fix
```
#define GPIO_AF9_CAN1 GPIO_AF9_CAN
#define CAN1_RX0_IRQn CAN_RX0_IRQn
#define CAN1_TX_IRQn CAN_TX_IRQn
#define GPIO_SPEED_FREQ_VERY_HIGH GPIO_SPEED_FREQ_HIGH
```

Unfortunately the chip hangs when sending a message. Specifically in 
`__HAL_CAN_ENABLE_IT(n_pCanHandle, CAN_IT_TX_MAILBOX_EMPTY);`

tyllj, 2024-02-24:
> Did a temporary fix
>
> ```
> #define GPIO_AF9_CAN1 GPIO_AF9_CAN
> #define CAN1_RX0_IRQn CAN_RX0_IRQn
> #define CAN1_TX_IRQn CAN_TX_IRQn
> #define GPIO_SPEED_FREQ_VERY_HIGH GPIO_SPEED_FREQ_HIGH
> ``` [...]

The F303 crashes because the ISRs are not defined correctly for parts with a single CAN interface.
With two additional defines this workaround actually worked for me. Here is my hal_conf_extra.h:
```c
#define HAL_CAN_MODULE_ENABLED

// Workaround for compatibility issue
#define GPIO_AF9_CAN1 GPIO_AF9_CAN
#define CAN1_RX0_IRQn CAN_RX0_IRQn
#define CAN1_TX_IRQn CAN_TX_IRQn
#define GPIO_SPEED_FREQ_VERY_HIGH GPIO_SPEED_FREQ_HIGH
#define CAN1_TX_IRQHandler CAN_TX_IRQHandler
#define CAN1_RX0_IRQHandler CAN_RX0_IRQHandler
```

pazi88, 2024-03-21:
> The F303 crashes because the ISRs are not defined correctly for parts with a single CAN interface. With two additional defines this workaround actually worked for me. Here is my hal_conf_extra.h: [...]

I couldn't get any better fix for this to work. So this is now included in the library. Test if it works.

closed 2024-03-21 by pazi88