pazi88/STM32_CAN #19: Stm32f1 is leaving sending the message

isarwar93, 2023-10-19:
Hello, 

I am using your library to communicate with ESP32-S3 using CAN bus. Sometimes it works really fine. However, most of the times, STM32F1 stops sending the data, I have to restart the STM32 to start the communication again. 

I put Can.begin(); in the loop just to check, discontinuity got much better, but still sometimes stops sending.

Do you know what could be the issue?

Kind Regards,
Ismail

isarwar93, 2023-10-19:
The problem is with my power supply i think, which is quite noisy. It gets better if I connect more grounds here and there.

But still what could be the issue that it stops sending the signal? Although, the loop is not getting stuck, (it keeps doing the serial printing in the loop). But the signals on the Can output pins are disappearing.

pazi88, 2023-10-30:
I have also noticed this. It seems to be bit of a random issue that has come recently. I haven't yet had time to investigate this, but I will.