pazi88/STM32_CAN #22: Support for STM32G0B1 with different CAN pin numbers, CAN2 on PB0/PB1

rondlh, 2023-11-25:
I'm trying to get CAN to work on a BTT EBB CAN board (https://github.com/bigtreetech/EBB) which is a 3D printer tool head board.
The version V1.2 board I have uses a STM32G0B1 MCU, which uses pin number for the CAN bus that are currently not supported by the library:

**STM32G0 CAN bus pin numbers: (All are AF3)**
CAN1_RX PA11
CAN1_TX PA12
**CAN2_RX PB0 (Used by BTT EBB42 V1.2)
CAN2_TX_PB1    (Used by BTT EBB42 V1.2)**
CAN2_RX PB5
CAN2_TX_PB6
CAN1_RX PA8
CAN1_TX PA9
CAN2_RX PB12
CAN2_TX_PB13
CAN2_RX PC2
CAN2_TX_PC3
CAN1_RX PC4
CAN1_TX_PC5

I could give this a try myself, some pointers in the right direction would be appreciated.

rondlh, 2023-11-25:
Would this do the trick? (Added in the CAN2 pin section).

```
    if (_pins == ALT_2) // CAN2 MAPPED TO PIN B0/B1 FOR STM32G0
    {
      __HAL_RCC_GPIOB_CLK_ENABLE();
      #if defined(__HAL_RCC_AFIO_CLK_ENABLE)
        __HAL_AFIO_REMAP_CAN2_ENABLE(); // To use PB0/1 pins for CAN2. Don't ask me why this has different name than for CAN1.
        __HAL_RCC_AFIO_CLK_ENABLE();
        GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
        GPIO_InitStruct.Pin = GPIO_PIN_0;          // CAN_RX, PB0
        GPIO_InitStruct.Mode = GPIO_MODE_INPUT;
        HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);
        GPIO_InitStruct.Pin = GPIO_PIN_1;          // CAN_TX, PB1
      #else
        GPIO_InitStruct.Alternate = GPIO_AF3_CAN2; // AF3
        GPIO_InitStruct.Pin = GPIO_PIN_0|GPIO_PIN_1;
        GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
      #endif
      GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
      HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);
    }
```

pazi88, 2024-04-11:
STM32G0B1 uses CAN-FD, which is not yet supported with this library. My goal is to get it working some day, but no ETA.

closed 2024-04-11 by pazi88