pazi88/STM32_CAN #24: STM32F405 is compiling but not Working

ToRoCu, 2024-01-06:
Hello Pazi, I wanted to use Can on an STM32F405, but unfortunately nothing works.
Compiling brings no errors or messages.
I have tested the example "write" with `STM32_CAN Can( CAN1, ALT );` but nothing works.
None of the GPIO's work (tested with oscilloscope).

What could be wrong?

igoro00, 2024-03-02:
Same issue but on the bluepill (STM32F103). I tested it in `CAN1, DEF` and `CAN1, ALT` mode and nothing shows up on the pins when testing with a logic analyzer.

Have you managed to solve the issue?

ToRoCu, 2024-03-03:
unfortunately not, something has changed again in the HAL commands which means that some libs no longer work. you can test with an older version of the Arduino board package 1.6, maybe you will have more luck then.

pazi88, 2024-03-11:
Just to confirm, that do you have CAN transceiver and wires with termination resistors connected?

igoro00, 2024-03-11:
> Just to confirm, that do you have CAN transceiver and wires with termination resistors connected?

I have and it didn't work. I switched to [ExoCAN](https://github.com/exothink/eXoCAN) and it worked so there's definitely something wrong with the library.

ToRoCu, 2024-03-11:
> Just to confirm, that do you have CAN transceiver and wires with termination resistors connected?

I tested directly without a tranceiver, after that didn't work I measured the pins with an oscilloscope and saw that there were no state changes. 

@igoro00 Thanks for the tip with _ExoCAN_ unfortunately this only serves the F103 otherwise I would have liked to test it.

igoro00, 2024-03-11:
> @igoro00 Thanks for the tip with _ExoCAN_ unfortunately this only serves the F103 otherwise I would have liked to test it.

Did you actually test it though? I've briefly looked at the datasheets for both f103 and f105. They both use the same pinout for can1(PA11 and PA12 as the default mode) and the memory locations are the same(again, for can1, because f103 only has one can interface). So unless the memory locations for setting up gpio and clocks are different(I doubt it) it should just work.

I'd say, try it. Maybe it'll just work.

pazi88, 2024-03-15:
> I tested directly without a tranceiver, after that didn't work I measured the pins with an oscilloscope and saw that there were no state changes.
>
> @igoro00 Thanks for the tip with _ExoCAN_ [...]

Yes this is normal in this implementation. There is nothing really happening on the pins if you don't have CAN transceiver and CAN bus with terminators there.

Anyways, I tried this current library version with the  library examples on my F103C8 and F407VE boards. Using the latest 2.7.1 STM32 core and Arduino IDE 2.3.0. And everything works as it should. I can't find any problems.

If you are seeing problems with F103, please tell the exact combo. Note that the CAN transceiver and CAN bus with termination resistors needs to be present.

ToRoCu, 2024-04-03:
Hi, in the meantime I got round to testing, took a closer look at what Pazi wrote and tried the whole thing again, after a few tries and finally a complete reinstall of the board packages it worked.
I think something got stuck when I tried the different packages, plus possibly contact issues on the breadboard at the beginning...
is supposed to happen
thanks to all

@igoro00 i hope your problem will also be solved so easily

closed 2024-04-03 by ToRoCu