pazi88/STM32_CAN #25: Problem with multiple CAN lines

siksndavis, 2024-02-02:
Hello, using STM32F103RBT6 board with two TJA1050 transceivers on it. 
When trying to compile code for first CAN bus line, everything goes smooth, I receive and send data, all goes well.
Problem starts when trying to compile code for any other CAN bus line (2/3/4), even using the example code.

`/*
This is simple example to read all data from CAN bus and print it out to serial bus.
*/

#include "STM32_CAN.h"
//STM32_CAN Can( CAN1, DEF );  //Use PA11/12 pins for CAN1.
//STM32_CAN Can( CAN1, ALT );  //Use PB8/9 pins for CAN1.
//STM32_CAN Can( CAN1, ALT_2 );  //Use PD0/1 pins for CAN1.
STM32_CAN Can2( CAN2, DEF );  //Use PB12/13 pins for CAN2.
//STM32_CAN Can( CAN2, ALT );  //Use PB5/6 pins for CAN2
//STM32_CAN Can( CAN3, DEF );  //Use PA8/15 pins for CAN3.
//STM32_CAN Can( CAN3, ALT );  //Use PB3/4 pins for CAN3

static CAN_message_t CAN_RX_msg;

void setup() {
  Serial.begin(115200);
  Can2.begin();
  //Can.setBaudRate(250000);  //250KBPS
  Can2.setBaudRate(500000);  //500KBPS
  //Can.setBaudRate(1000000);  //1000KBPS
}

void loop() {
  if (Can2.read(CAN_RX_msg) ) {
    Serial.print("Channel:");
    Serial.print(CAN_RX_msg.bus);
    if (CAN_RX_msg.flags.extended == false) {
      Serial.print(" Standard ID:");
    }
    else {
      Serial.print(" Extended ID:");
    }
    Serial.print(CAN_RX_msg.id, HEX);

    Serial.print(" DLC: ");
    Serial.print(CAN_RX_msg.len);
    if (CAN_RX_msg.flags.remote == false) {
       Serial.print(" buf: ");
      for(int i=0; i<CAN_RX_msg.len; i++) {
        Serial.print("0x"); 
        Serial.print(CAN_RX_msg.buf[i], HEX); 
        if (i != (CAN_RX_msg.len-1))  Serial.print(" ");
      }
      Serial.println();
    } else {
       Serial.println(" Data: REMOTE REQUEST FRAME");
    }
  }
}`

`C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved202412-17680-d3dugb.57k8k\Read\Read.ino:9:17: error: 'CAN2' was not declared in this scope; did you mean 'Can2'?
    9 | STM32_CAN Can2( CAN2, DEF );  //Use PB12/13 pins for CAN2.
      |                 ^~~~
      |                 Can2

exit status 1

Compilation error: 'CAN2' was not declared in this scope; did you mean 'Can2'?`

And thats like that all the time. CAN1 works smooth. Can you help?

pazi88, 2024-02-02:
As far as I know F103 only has single CAN imterface that has quite many alternative pins. But that's why CAN2 doesn't work, because it doesn't have it.

siksndavis, 2024-02-02:
Thanks man, you were right! I'm still noob at MCU's :)
Will try with other board I have, STM32F105RBT6, but I just tried to compile for test with F205 and it compiled like a charm!

closed 2024-03-11 by pazi88