pazi88/STM32_CAN #30: Filter logic is wrong

ppcd9220, 2024-03-23:
When setting filters there is no way to set it for Standard or Extended ID.

What type of messages to filter is defined here:
`  if (filter_id <= 0x7FF)
  {
    // Standard ID can be only 11 bits long

  }`
This is completely wrong logic:
If the ID less than 0x7FF - Standard ID, otherwise - EXT.

i.e Message with ID= 0x01 Could be both EXT and Standard.
But with current approach all messages less than 0x7FF and EXT ID flag active is not possible to pass the filter.

pazi88, 2024-03-26:
hmmnn, yeah. Need to fix that.

ppcd9220, 2024-03-26:
> hmmnn, yeah. Need to fix that.
Maybe just add a flag: EXT_ID ?

pazi88, 2024-03-27:
> > hmmnn, yeah. Need to fix that.
> > Maybe just add a flag: EXT_ID ?

Looks like that's the thing I need to do. At least for me it looks like that you need to specify that is the ID standard or extended. I can't seem to make it accept both. So I'll add a flag to set that.

closed 2024-03-27 by pazi88

ppcd9220, 2024-03-27:
> > > hmmnn, yeah. Need to fix that.
> > > Maybe just add a flag: EXT_ID ?
> 
> Looks like that's the thing I need to do. At least for me it looks like that you need to specify that is the ID standard or extended. I can't seem to make it accept both. So I'll add a flag to set that.

Yes. If you need to accept both - you can setup 2 different filters: one for standard and one for EXT.
This flag (Ext_ID) in CAN_ID completely separates address spaces for Standard and Extended. And they don't cross or interact. That's why it is absolutely safe to work at the same time with Can_Open messages/devices that are using 11-bit standard addressing and J1939 that uses EXT (29-bit) IDs

ppcd9220, 2024-03-27:
> So I'll add a flag to set that.

Thank you Very Much for your Work on this lib!

commit da257996 (2024-03-27): Add possibility to specify ID format for filters.

Fixes #30