pazi88/STM32_CAN #32: Nucleo-64 F072RB compiling error: 'GPIO_AF9_CAN1' was not declared in this scope; did you mean 'GPIO_AF4_CAN'?

DadoKomar, 2024-03-24:
Hello Pasi

I have found your library and have some old Nucleo, so decided to try if CAN works over Arduino.  First of all thanks for good job, interesting there is no such a library for STM32duino.

I have F072, not running USB. During compiling getting following error. I did a quick dive, just lost as dont understand why it is erroring in ALT_2 statement when I constructing STM32CAN with ALT  (wanna use PB8/PB9). Could you please look at it, I am almost sure you would have a fix for this. I just opened your Sniff example.

[image]

Here is the line 139, magic for me. ;)

[image]

And this is my setup of Ardu

[image]

Many thanks for any advice and your time.

Regards,
Dado

pazi88, 2024-03-26:
This should be fixed now.

DadoKomar, 2024-03-26:
I knew it will be a peace of cake for you. ;) Will try. Many thanks for great support.

pazi88, 2024-04-11:
I assume this is ok, so I'll close the issue.

closed 2024-04-11 by pazi88