pazi88/STM32_CAN #34: Hangs on STM32F042

TT-ZX, 2024-04-21:
I am getting a hang after calling can.write().  If I comment out line 330 `__HAL_CAN_ENABLE_IT(n_pCanHandle, CAN_IT_TX_MAILBOX_EMPTY);` in STM32_CAN.cpp then I no longer get the hang.  This was also reported in issue #14 .

moksalahti, 2024-07-02:
I have the same issue. I'm using the latest version and even though in issue #14 this is claimed to have been fixed, it looks like it's not. Commenting out the line with __HAL_CAN_ENABLE_IT(n_pCanHandle, CAN_IT_TX_MAILBOX_EMPTY); does prevent the hang but I get a similar right away when I try to do a Can.read().

TT-ZX, 2024-07-05:
Yeah, I'm having the same issue.  I didn't notice this at first because my application was only sending CAN packets, not receiving.

closed 2024-08-21 by TT-ZX