pazi88/STM32_CAN #36: Support for PB8/PA12 CAN pin mapping on STM32F042

sshsphere, 2024-05-21:
It would be nice if this library natively supported use of PB8 (CAN RX) and PA12 (CAN TX) for the CAN interface.

This code generated from CubeIDE might be helpful:
```cpp
__HAL_RCC_GPIOB_CLK_ENABLE();
__HAL_RCC_GPIOA_CLK_ENABLE();
GPIO_InitStruct.Pin = GPIO_PIN_8;
GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
GPIO_InitStruct.Pull = GPIO_NOPULL;
GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
GPIO_InitStruct.Alternate = GPIO_AF4_CAN;
HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

GPIO_InitStruct.Pin = GPIO_PIN_12;
GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
GPIO_InitStruct.Pull = GPIO_NOPULL;
GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
GPIO_InitStruct.Alternate = GPIO_AF4_CAN;
HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);
```
I added this option to the begin function in `STM32_CAN.cpp` and everything works perfectly

pazi88, 2024-05-21:
Ok. Good to know that it works. I'll see what I can do. Some use different way of setting up the pins that are used and in those you can only use the pin pairs that are currently available in the library. I need to think best way to do that.

pazi88, 2025-01-23:
Closed via #45

closed 2025-01-23 by pazi88