pazi88/STM32_CAN #37: Using CAN1 and CAN2 simultaneously only works by modifying the initializeFilters() function.

tscheskn, 2024-05-28:
Hello,
I'm using both CAN1 and CAN2 on a STM32F446RE. CAN1 works fine.
But with the default initializeFilters() functions that is called in setBaudRate(), CAN2 was only able to send but not receive messages. 
The fix I found was also setting the SlaveStartFilterBank to 14 for CAN2:

```
void STM32_CAN::initializeFilters() {
  CAN_FilterTypeDef sFilterConfig;
  // We set first bank to accept all RX messages
  sFilterConfig.FilterBank = 0;
  sFilterConfig.FilterMode = CAN_FILTERMODE_IDMASK;
  sFilterConfig.FilterScale = CAN_FILTERSCALE_32BIT;
  sFilterConfig.FilterIdHigh = 0x0000;
  sFilterConfig.FilterIdLow = 0x0000;
  sFilterConfig.FilterMaskIdHigh = 0x0000;
  sFilterConfig.FilterMaskIdLow = 0x0000;
  sFilterConfig.FilterFIFOAssignment = CAN_RX_FIFO0;
  sFilterConfig.FilterActivation = ENABLE;
#ifdef CAN2
  // Filter banks from 14 to 27 are for Can2, so first for Can2 is bank 14. This
  // is not relevant for devices with only one CAN
  if (_canPort == CAN1) {
    sFilterConfig.SlaveStartFilterBank = 14;
  }
  if (_canPort == CAN2) {
    sFilterConfig.FilterBank = 14;
    sFilterConfig.SlaveStartFilterBank = 14; // SlaveStartFilterBank also needs to be set for CAN2 apparently
  }
#endif

  HAL_CAN_ConfigFilter(n_pCanHandle, &sFilterConfig);
}
```

I'm not sure if this is a special issue with my setup (I can provide more details if necessary) or a general issue.

pazi88, 2024-05-28:
Could be common issue. I should soon have my new board in hands that has both CAN1 and CAN2 available, so I can test this too and include the fix in library, if needed

pazi88, 2025-01-23:
Closed via #45

closed 2025-01-23 by pazi88