pazi88/STM32_CAN #38: Not Stable when using FreeRTOS

wltue, 2024-07-05:
Haai

I cannot seem to get it stable while using STM32FreeRTOS.
It seems to work when i only send on a empty bus (only a Kvaser listening device connected)
But when adding more nodes and busload it seems to crash randomly.
Using a Nucleo STM32F303RE board with a MCP2542FD chip.

Works fine when i use https://github.com/nopnop2002/Arduino-STM32-CAN with STM32 FreeRTOS on the same hardware.

Thank you.

Included project:
[can4.zip](https://github.com/user-attachments/files/16107506/can4.zip)

wltue, 2024-07-05:
Seems to execute:

/*-----------------------------------------------------------*/
#if ( configCHECK_FOR_STACK_OVERFLOW >= 1 )
	/**  Blink three short pulses if stack overflow is detected.
	Run time stack overflow checking is performed if
	configCHECK_FOR_STACK_OVERFLOW is defined to 1 or 2.  This hook
	function is called if a stack overflow is detected.
  \param[in] pxTask Task handle
  \param[in] pcTaskName Task name
  */
void vApplicationStackOverflowHook(TaskHandle_t pxTask, char *pcTaskName) {
  (void) pcTaskName;
  (void) pxTask;
  errorBlink(3);
}
#endif /* configCHECK_FOR_STACK_OVERFLOW >= 1 */

But why @ recieving?

mehrdad987, 2024-07-08:
Set the recovery time to less than 500ms