pazi88/STM32_CAN #39: The example does not compile on nucleo_f429zi

chriskinal, 2024-07-24:
The example does not compile under the Arduino framework Ends with the following errors;

In file included from src/main.cpp:7:
.pio/libdeps/nucleo_f429zi/STM32_CAN/STM32_CAN.h:191:33: error: 'CAN_HandleTypeDef' has not been declared
  191 |     void      calculateBaudrate(CAN_HandleTypeDef *CanHandle, int Baudrate);
      |                                 ^~~~~~~~~~~~~~~~~
.pio/libdeps/nucleo_f429zi/STM32_CAN/STM32_CAN.h:192:33: error: 'CAN_HandleTypeDef' has not been declared
  192 |     void      setBaudRateValues(CAN_HandleTypeDef *CanHandle, uint16_t prescaler, uint8_t timeseg1,
      |                                 ^~~~~~~~~~~~~~~~~
Compiling .pio/build/nucleo_f429zi/lib311/STM32duino LwIP/core/ipv4/icmp.c.o
.pio/libdeps/nucleo_f429zi/STM32_CAN/STM32_CAN.h:241:5: error: 'CAN_HandleTypeDef' does not name a type; did you mean 'SAI_HandleTypeDef'?
  241 |     CAN_HandleTypeDef *n_pCanHandle;
      |     ^~~~~~~~~~~~~~~~~
      |     SAI_HandleTypeDef
Compiling .pio/build/nucleo_f429zi/lib311/STM32duino LwIP/core/ipv4/igmp.c.o
.pio/libdeps/nucleo_f429zi/STM32_CAN/STM32_CAN.h:151:108: error: 'CAN_FILTERMODE_IDMASK' was not declared in this scope
  151 |     bool setFilter(uint8_t bank_num, uint32_t filter_id, uint32_t mask, IDE = AUTO, uint32_t filter_mode = CAN_FILTERMODE_IDMASK, uint32_t filter_scale = CAN_FILTERSCALE_32BIT, uint32_t fifo = CAN_FILTER_FIFO0);
      |                                                                                                            ^~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/nucleo_f429zi/lib311/STM32duino LwIP/core/ipv4/ip4.c.o
Compiling .pio/build/nucleo_f429zi/lib311/STM32duino LwIP/core/ipv4/ip4_addr.c.o
.pio/libdeps/nucleo_f429zi/STM32_CAN/STM32_CAN.h:151:155: error: 'CAN_FILTERSCALE_32BIT' was not declared in this scope
  151 |     bool setFilter(uint8_t bank_num, uint32_t filter_id, uint32_t mask, IDE = AUTO, uint32_t filter_mode = CAN_FILTERMODE_IDMASK, uint32_t filter_scale = CAN_FILTERSCALE_32BIT, uint32_t fifo = CAN_FILTER_FIFO0);
      |                                                                                                                                                           ^~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/nucleo_f429zi/STM32_CAN/STM32_CAN.h:151:194: error: 'CAN_FILTER_FIFO0' was not declared in this scope; did you mean 'CAN_FilterFIFO0'?
  151 |     bool setFilter(uint8_t bank_num, uint32_t filter_id, uint32_t mask, IDE = AUTO, uint32_t filter_mode = CAN_FILTERMODE_IDMASK, uint32_t filter_scale = CAN_FILTERSCALE_32BIT, uint32_t fifo = CAN_FILTER_FIFO0);
      |                                                                                                                                                                                                  ^~~~~~~~~~~~~~~~
      |                                                                                                                                                                                                  CAN_FilterFIFO0

itmmckernan, 2024-10-06:
Did you add the proper build flags to enable CAN in the HAL? Adding `-D HAL_CAN_MODULE_ENABLED` to the build_flags in my platformio.ini fixed this for me.

chriskinal, 2024-10-07:
My apologies for not closing this issue in a more timely manner. Someone on the PlaformIO forum pointed me in the right direction and I got it to work.

closed 2024-10-07 by chriskinal