pazi88/STM32_CAN #40: Baudrate is calculated with SJW in the mix

attie-argentum, 2024-07-24:
We've been having some issues with CAN communications between a handful of devices.
I started looking in to how the baudrate is calculated, and it seems to be slightly wrong.

`SJW` is not actually part of the bit-time calculation, I suspect this has been confused with `SYNC_SEG` (which is always `1`).
The following is from the STM32F72xxx reference manual, but applies to all bxCAN peripherals.

[image]

I believe the calculation below should be `calcBaudrate = (frequency / (prescaler * (1 + bs1 + bs2)))`.

https://github.com/pazi88/STM32_CAN/blob/ae6ccadd8d39d245fbddee38bec2206114c481ab/STM32_CAN.cpp#L707-L717

Separately, to improve interoperability SJW can be increased from the default value of `1` - this will allow the next transition to arrive earlier than the node expects it - up to `min(BS2, 4)` time quanta.
In many cases, this can be safely set higher than `1` - and in other cases it's even necessary to avoid error frames and wider disruption on the bus.

Please can you let me know if you agree with this, or if you think I'm mistaken?
I'd be happy to prepare a PR to adjust this accordingly.

pazi88, 2024-07-26:
You might be right. Please prepare the PR. I'll check it more in detail next week.

pazi88, 2024-08-21:
Fixed in #41

closed 2024-08-21 by pazi88