pazi88/STM32_CAN #43: Will not compile for STM32F407VET6

Kody68, 2024-09-01:
Will not compile due to errors below. They all show to be in the STM32_CAN library.

error: 'CAN_HandleTypeDef' has not been declared
  194 |     bool      lookupBaudrate(CAN_HandleTypeDef *CanHandle, int Baudrate, const T(&table)[N]);
error: 'CAN_FILTERMODE_IDMASK' was not declared in this scope
similar error on lines 195 and 196

error: 'CAN_FILTERMODE_IDMASK' was not declared in this scope
  152 |     bool setFilter(uint8_t bank_num, uint32_t filter_id, uint32_t mask, IDE = AUTO, uint32_t filter_mode = CAN_FILTERMODE_IDMASK, uint32_t filter_scale = CAN_FILTERSCALE_32BIT, uint32_t fifo = CAN_FILTER_FIFO0);
error: 'CAN_FILTERSCALE_32BIT' was not declared in this scope
  152 |     bool setFilter(uint8_t bank_num, uint32_t filter_id, uint32_t mask, IDE = AUTO, uint32_t filter_mode = CAN_FILTERMODE_IDMASK, uint32_t filter_scale = CAN_FILTERSCALE_32BIT, uint32_t fifo = CAN_FILTER_FIFO0);
error: 'CAN_FILTER_FIFO0' was not declared in this scope; did you mean 'CAN_FilterFIFO0'?
  152 |     bool setFilter(uint8_t bank_num, uint32_t filter_id, uint32_t mask, IDE = AUTO, uint32_t filter_mode = CAN_FILTERMODE_IDMASK, uint32_t filter_scale = CAN_FILTERSCALE_32BIT, uint32_t fifo = CAN_FILTER_FIFO0);

error: 'CAN_HandleTypeDef' does not name a type; did you mean 'DAC_HandleTypeDef'?
  245 |     CAN_HandleTypeDef *n_pCanHandle;

Any idea what could be causing this issue?

pazi88, 2024-09-02:
Have you enabled the HAL CAN module on build parameters, or do you have the [hal_conf_extra.h](https://github.com/pazi88/STM32_CAN/blob/main/examples/Read/hal_conf_extra.h) file there with the code to enable it.

pazi88, 2024-11-27:
I'll close because this is most likely user error.

closed 2024-11-27 by pazi88