pazi88/STM32_CAN #44: "CAN_HandleTypeDef has not been declared"

SpeedieKoala, 2024-11-15:
The output terminal throws many errors but it all starts with "CAN_HandleTypeDef has not been declared", most variables that include Can or HAL in the name also have an error associated with them. We are using this on platform io with arduino and STM32F09. The errors are all based in the library. We are running it with the examples provided in the [Read-Write_with_filters] folder. (https://github.com/pazi88/STM32_CAN/tree/main/examples/Read-Write_with_filters)

jlaustill, 2024-11-15:
I'm getting these same errors today. And all I've done is added this as a lib and tried to import it. 

platformio.ini
```
[env:weact_mini_h743vitx]
platform = ststm32
board = weact_mini_h743vitx
framework = arduino
upload_protocol = stlink
debug_tool = stlink
monitor_speed = 115200
lib_deps =
	git@github.com:adafruit/Adafruit_ADS1X15.git
	git@github.com:pazi88/STM32_CAN.git
    SPI
	jlaustill/sea-dash
```
And my super simple main.cpp :) 
```
#include <Arduino.h>
#include <Wire.h>
#include <Adafruit_ADS1X15.h>
#include <SeaDash.hpp>
#include <cmath>

#include <STM32_CAN.h>

Adafruit_ADS1115 ads1115;
TwoWire I2C1_Custom(PB9, PB8);

uint32_t lastBlink = 0;
uint8_t LED_PIN = PA1;
bool adsSuccess = false;

void setup() {
    Serial.begin(115200);
    pinMode(LED_PIN, OUTPUT);
    digitalWrite(LED_PIN, 0);
    lastBlink = millis();
    I2C1_Custom.begin();
    adsSuccess = ads1115.begin(0x48, &I2C1_Custom);
    if (!adsSuccess) {
        Serial.println("Failed to init ads!");
    }
}

void loop() {
    uint32_t currentMillis = millis();
    int16_t adc0;

    if ((currentMillis - lastBlink) >= 1000) {
        lastBlink = currentMillis;
        digitalWrite(LED_PIN, !digitalRead(LED_PIN));
        Serial.println("blinky blink " + (String) LED_PIN);
        if (adsSuccess) {
            adc0 = ads1115.readADC_SingleEnded(0);
            float volts = ads1115.computeVolts(adc0);
            //float psia = 25.00f * (volts- 0.5f);
            float psia = SeaDash::Floats::mapf(volts, 0.5f, 4.5f, 0.0f, 150.0f);
            Serial.print("AIN0: "); Serial.print(volts); Serial.print(" "); Serial.println(psia);
        } else {
            Serial.println("ads1115 failed :(");
        }
    }
}
```
Any direction on debugging this would be fantastic!

pazi88, 2024-11-15:
Usually "CAN_HandleTypeDef has not been declared" build error means that the MCU doesn't have CAN interface, or the CAN interface is the newer CAN-FD one, that isn't supported yet. The h743 has CAN-FD, so it's clear why this library doesn't compile for it. But F09 I'm not sure. Quick look on ST's documentation doesn't tell which CAN interface it has.

vaaniamkhan, 2024-11-16:
I'm working on the F09 with @SpeedieKoala. 

We went through the doc from STM32 that details the implementation of FD and appears only the G0, G4, H5, H7, L5, and U5 series have it, not F0. Documentation about ours specifically only mentions CAN 2.0A and B active.

Could there be a different reason for the problem?

pazi88, 2024-11-27:
What software/stm32duino version you use? I did quickly try the examples as is for F09 and F103RET, and those compile just fine.

jmanncontrel, 2024-11-27:
PlatfomIO v3.3.3 and STM32 v17.6.0 but I found the issue. Instead of enabling the CAN module via hal_conf_extra.h, I added "-DHAL_CAN_MODULE_ENABLED" to the build flags in platformio.ini. It now compiles without any errors.

pazi88, 2024-11-27:
> PlatfomIO v3.3.3 and STM32 v17.6.0 but I found the issue. Instead of enabling the CAN module via hal_conf_extra.h, I added "-DHAL_CAN_MODULE_ENABLED" to the build flags in platformio.ini. It now compiles without any errors.

Yes. It's on the instructions that it needs to be added for PIO. I'll close this issue.

jmanncontrel, 2024-11-27:
No problem - that's where I spotted it. My mistake.