pazi88/STM32_CAN #46: 1.2.0

DmitryKand, 2025-03-03:
Please bring the library (1.2.0) to a working state or remove it from auto-update.

pazi88, 2025-03-03:
What is the problem? I did do quite a bit of testing for the current version and everything worked.

DmitryKand, 2025-03-03:
this code on (1.1.3) works fine, after the update it doesn’t work at all.

DmitryKand, 2025-03-03:
#include <STM32_CAN.h>

static CAN_message_t CAN_inMsg;
static CAN_message_t CAN_outMsg;

//STM32_CAN Can1(PA_11, PA_12, RX_SIZE_256, TX_SIZE_256);
STM32_CAN Can1(CAN1, DEF, RX_SIZE_256, TX_SIZE_256);

void setup() {

#ifdef debugging
  Serial.begin(115200);
#endif
  Can1.begin();
  Can1.setBaudRate(95238); 

  /*Can1.setFilterSingleMask(MB0, 0x180, 0x7FF, STD);  //1 sec
  Can1.setFilterSingleMask(MB1, 0x206, 0x7FF, STD);
  Can1.setFilterSingleMask(MB2, 0x450, 0x7FF, STD);
  Can1.setFilterSingleMask(MB3, 0x4EC, 0x7FF, STD);
  Can1.setFilterSingleMask(MB4, 0x4E8, 0x7FF, STD);
  Can1.setFilterSingleMask(MB5, 0x506, 0x7FF, STD);  //0.2 sec*/

  Can1.setMBFilterProcessing(MB0, 0x180, 0x7FF);  //1 sec
  Can1.setMBFilterProcessing(MB1, 0x206, 0x7FF);
  Can1.setMBFilterProcessing(MB2, 0x450, 0x7FF);
  Can1.setMBFilterProcessing(MB3, 0x4EC, 0x7FF);
  Can1.setMBFilterProcessing(MB4, 0x4E8, 0x7FF);
  Can1.setMBFilterProcessing(MB5, 0x506, 0x7FF);  //0.2 sec

void loop() {}

pazi88, 2025-03-03:
What do you expect it do do? That just sets the filters and CAN baud rate to really weird one?

DmitryKand, 2025-03-03:
Well, of course this is not the whole code.

pazi88, 2025-06-10:
Tested the current library version on multiple STM32 processors and all seems ok. I'll close this.

closed 2025-06-10 by pazi88