pazi88/STM32_CAN #47: Error using bulepill(STM32F103C8T6) in Arduino IDE

Allen1080211, 2025-03-03:
I downloaded two Bluepills and used the MCP2551 using the example files Write and Read respectively, but I cannot get the output string in the serial monitor of the Arduino IDE. what can I do?

[image]

[image]

pazi88, 2025-03-03:
As it says in the error. The USB and CAN share interrupts, so you can't use USB and CAN simultaneously. You need use HW serial with CAN.

Allen1080211, 2025-03-03:
I can't select STM32_CAN Can( CAN1, ALT ); //Use PB8/9 pins for CAN1?

Please help me Thanks.

duckylotl, 2025-03-10:
In your seconds screenshot it looks like your HardwareSerial on your receiver is miss-configured.
`HardwareSerial Serial2(PB3, PB2)`

PB2 is dual used with the BOOT1 pin and usually not really intended to be used for much else. Also neither PB2 nor PB3 have any UART functionality, so the HardwareSerial will not work.

This seems like a typo since your sending implementation uses (PA3, PA2) which in fact have UART2 hardware connected to it.
Alternative pin configs would be (PB11, PB10) for UART3. And either (PB7, PB6) or (PA10, PA9) for UART1.

Consider printing a fixed message on start to confirm to yourself that your UART is in fact working as expected.

Allen1080211, 2025-03-14:
I can't follow the example code to set STM32_CNA Can( CAN1, ALT ); to connect two STM32 CAN? Because it still doesn't work.

Serial2 is just because I want to check the data through the USB to TTL module.

duckylotl, 2025-03-15:
You use Serial2 as your method of verification of the CAN path (debug messages).
If Serial2 is broken, you can not claim that CAN is at fault. It still might be broken, but current evidence can not claim only CAN is broken.

Please verify that your test messages can actually be delivered before claiming that CAN is the only fault.
Or do you have other means to signal successful / non-successful reception of CAN frames to you that does not rely on the Serial connection (eg. status LED)?

What would also help is a scope or logic probe to see if the Tx/Rx lines of the can transceivers and/or the H/L lines of the the CAN bus are carrying your expected messages. This would give you a hint if the problem is in the transmission or the reception.

Cheap clones of the Saleae logic Analyzers can be found on places like ebay and are fast enough to capture this.

pazi88, 2025-06-10:
Tested current version on Arduino IDE and should be ok now.

closed 2025-06-10 by pazi88