pazi88/STM32_CAN #54: Fake STM32F103 deficiencies

duckylotl, 2025-04-28:
This is less of a bugreport for the driver, but to document a problem with fake STM32F103 that are distributed on popular cheap "BluePill" boards.

I recently bought a new batch and it turns out I got fake ICs that are in fact CKS CS32F103C8T6 (even though they carry an STM logo).
This is noticeable since the DAP idcode returns `0x2ba01477` instead of the expected `0x1ba01477` for the STM one.

In general the fake one functions similar to the original. Looking in detail at the CAN functionality though, there are differences.

### CODE field of TSR Register is defective
One mayor thing is that the CODE field in the TSR register that is supposed to give the driver the address of the next free messagebox seems to be defective. At lest if it is supposed to behave like the original STM one.
The HAL driver uses this value to select the messagebox to write the next message.

From STM documentation its supposed to start a 0 by default. Once the 0th messagebox is filled it changes to the next free address (1) to signal that the next message should be written there.
Sidenote: If all 3 messageboxes are full, the CODE points to the messagebox holding the message with the lowest priority. Presumably this could be used to evict a low priority message for a new important message, this is not use in the HAL nor in this driver though.

This trace shows the `CODE` value on trace D2 (CODE0) and D3 (CODE1) while transmitting 5 messages. Each time a message was written by this driver a pin was toggled (trace D0). The first 3 messages are ending up in the 3 hardware mailbox buffers, the other 2 end up in software fifo.

[image]

As expected CODE starts with a value of 0 and each time a message has been written it increases by 1, ending at 2 once every messagebox is full. The HAL function `HAL_CAN_AddTxMessage` used by this driver checks if any messageboxes are full before using the `CODE` value to select a mailbox. Since everything is full the remaining messages are put into the drivers software FIFO to be written into the mailboxes once there is space again.

The following shows the same firmware on the fake one:

[image]
The `CODE` value somehow starts at a value of `2`. This is unexpected but not really critical. Once the first message was written it also changes to a different buffer address (1). However once the 2nd message was written the value does not change anymore.
So even though the 2nd and 1st messageboxes are now full, it reports that the next mailbox to use is again the first one.

The `TME[0:2]` bits indicating which messagebox is empty do correctly report that only messagebox 0 is free (not shown in graph, tested separately).
This causes the HAL function `HAL_CAN_AddTxMessage` for the next 3 writes to detect that there is still one free messagebox (MB0) but then write all of the remaining messages to MB1 as indicated by `CODE`. This overwrites the 2nd message with the messages 3-5, leaving only message 5 at the end and a loss of messages 2-4.
From STM documentation we should loose write access to mailboxes once their TXRQ bit is set. For the fake IC this does not seem to be enforced, however I did not test if genuine F103 do enforce this either.

In my test baudrate was fairly slow (125k), so all buffer writes happened during the transmit of messagebox 2 (first message), this could possibly also cause corrupted messages if the messagebox was overwritten during the transmission process.

### Workaround
I edited the HAL function `HAL_CAN_AddTxMessage` in file stm32f1xx_hal_can.c to work around this.

By replacing the mailbox selection from this
```c
/* Select an empty transmit mailbox */
transmitmailbox = (tsr & CAN_TSR_CODE) >> CAN_TSR_CODE_Pos;
```
to this
```c
/* Select an empty transmit mailbox */
if ((tsr & CAN_TSR_TME0) != 0U)
  transmitmailbox = 0;
else if ((tsr & CAN_TSR_TME1) != 0U)
  transmitmailbox = 1;
else
  transmitmailbox = 2;
```
solves the loss of packages.
However this likely causes problems with the Transmit FIFO priority and likely brakes the FIFO / transmit chronologically mode, which is set by default by this driver.

### Conclusion
CAN periphery does seem to behave differently on the fake CS32F103C8T6, mayor problems can be worked around but still not ideal.
I do not think this driver should support these fakes, I just wanted to list my findings in case anybody else was unlucky as I was and wants to get basic functionality back. But if you need a reliable CAN peripherals seek out the real STM32F103.

### Extra observation
In the two images above I also highlighted the time it takes from first buffer write to first edge on the CAN peripheral.
On the fake ones it takes `5-7µs`, while on the real ones it only takes `1-3µs`.
Also the time it takes to write a message does seem to different between the two. On the fake ones it is quicker and only takes `5µs`, while on the real ones it almost takes `9µs`

pazi88, 2025-04-29:
Interesting. I have also came across some STM32 clones that don't seem to have working CAN interface (or some other weird issues). But I haven't really dug into why those fail to work properly.

duckylotl, 2025-05-04:
Looks like you use the 1.1.2 version of the STM_CAN lib, so the transmit error should not be caused by the bug introduced in 1.2.0 regarding alternate pin mapping on the F1 platform.

In my case transmit works, but with potential data loss because of incorrect buffer handling. So your issue seems a bit different.

My workaround posted above would have to apply to the stm HAL, no idea if that is possible in the platformIO environment. Maybe needs a custom platform to be setup?

For reference, using `st-info --probe` from the `stlink-utils` package a genuine STM32103C8 reports as:
```
Found 1 stlink programmers
  version:    V2J37S7
  serial:     16004A002933353739303541
  flash:      65536 (pagesize: 1024)
  sram:       20480
  chipid:     0x410
  dev-type:   STM32F1xx_MD
```
while my fake ones say this:
```
Found 1 stlink programmers
  version:    V2J37S7
  serial:     16004A002933353739303541
  flash:      262144 (pagesize: 2048)
  sram:       65536
  chipid:     0x414
  dev-type:   F1xx_HD
```
The STM32CubeProgrammer also reports the `chipid` value as `Device ID` in the bottom right box with "Target information".

From reference manual:
> This field indicates the device ID.For low-density devices, the device ID is 0x412
> For medium-density devices, the device ID is 0x410
> For high-density devices, the device ID is 0x414
> For XL-density devices, the device ID is 0x430
> For connectivity devices, the device ID is 0x418

The important part is the chipid and decoded dev-type. The real one indicates a medium density (MD) device. While the fake one has a chipid for high-density devices. STM does not have any devices in the LQFP48 Package (STM32F1xxCx) with a high density memory layout. An STM part number would have to be something like: STM32F103CC (256k flash) or STM32F103CD (384k flash) or STM32F103CE (512k flash). Even weirder is that is pretends to have 64k RAM and 256k of flash, this combination is not available from STM in the complete 103 series.
I did not test if the fake devices actually has the extra memory it promotes.

Your `STM32F103C6` is a low density device and should report an id of `0x412`.
You could use one of the above tools to check what your chip reports and see if it does report an unexpected value or odd memory layout to test if you do have a fake in hand.

There are quite a few replicas of the 103 series, and some of those are unfortunately re-labeld as STM devices. On this page someone de-capped and compared the dies of these chips, here you can see that they are in fact different silicon and not just a copy of the original design, explaining the functional differences.
Link: https://www.richis-lab.de/STM32.htm (german, [google-translated to EN](https://www-richis--lab-de.translate.goog/STM32.htm?_x_tr_sl=de&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=sc))

gordonthree, 2025-05-04:
> My workaround posted above would have to apply to the stm HAL, no idea if that is possible in the platformIO environment. Maybe needs a custom platform to be setup?

I found the stm32f1xx_hal_can.c file buried in a hidden PIO directory under my home directory and after backing it up, made your suggested modification. It did not have any effect on my transmit problem. It was a long shot. I agree with your assessment, my issue is likely a different problem. 

I ran ``st-info --probe`` against my devices, it seems to match up with what was advertised:
```
Found 1 stlink programmers
  version:    V2J39S7
  serial:     6068FF286068FF286068FF28
  flash:      32768 (pagesize: 1024)
  sram:       10240
  chipid:     0x0412
  descr:      F1 Low-density device
```

Nice micro-photography of those die shots on Richi's website. A common theme with the counterfeits seems to be poor quality labeling. The labels on my two boards are crisp, and the surface of the chip itself is smooth, no texture like it was sanded and re-lasered or whatever it is they do.

Diving deeper into fake chips, it's likely my CAN transceivers are also counterfeit, so I could be fighting issues there as well.

Thanks for the response!

duckylotl, 2025-05-05:
Yea my fake one also has poor labeling, so it sounds like you have a genuine one.

Out of curiosity I checked out your linked code and deployed it onto my genuine STM32F103C8, (minor modification of commenting a switch case for `ACK_SWITCHBOX` out because its was not defined.) I also kept the platform setting as is, since my MCU only has more memory and no other differences the binary should work fine on the C8 as well.
And at first it also did not work.

It turns out that it was a power problem in my case. My programmer outputs enough power to run the board, but without the USB power connected it did not outupt anything on the can IOs.
The 3.3 rail was just at 2.9V when only supplied by the programmer, and 3.3V when supplied with the USB port.
So I think in general you code should be fine. Unless there is a weird difference between my STM32F103C**8** and your STM32F103C**6**.

After a while I did get into a state where the BOFF (bus off) flag was set. Then transmission will not work anymore. Unfortunately the error state discovery and recovery is not really covered by the STM32_CAN library.
If you get the same error you can try to enable the auto bus off recovery by calling the following after setBaudrate()
```
CAN1->MCR |= CAN_MCR_ABOM;
```
Then the periphery should try to recover from a bus off state on its own, but that's not 100% reliable.

With version 1.2 of the lib you can now re-start the can library at least to recover from errors (`end()` followed by re-init of can), yet discovery of errors is not implemented yet either. However version 1.2 also has an error with remapping the CAN on F1 platform, this should be fixed in my recent RP but not yet part of any release. That fix involves an extra define that needs to be set, so since you use platformio you would need to add the define `AFIO_MAPR_CAN_REMAP1` in the `platformio.ini`

gordonthree, 2025-05-05:
I always overlook common causes of problems and jump to suspecting something more obscure. The 3.3v rail at the transceiver on my breadboard with two blue pills is only 1.6v. That is two dev boards plus two transceivers, all powered by the STLink programmer. I am going to reinforce the power supply for the board and see what happens.

Thanks for the tip on enabling bus recovery.