pazi88/STM32_CAN #58: Can this be used on the STM32G431?

ShiPC123, 2025-08-15:
I am running this example on the STM32G431, but the compilation fails. Is there any plan to adapt this library function for the STM32G431?

[image]

pazi88, 2025-08-20:
> This library does currently only support the BxCAN peripherals. The G4 series uses the FDCAN peripheral.
> 
> I did update the library to support the FDCAN peripheral in [my fork (Branch: FDCAN)](https://github.com/duckylotl/STM32_CAN/tree/FDCAN). I did not get around to fully clean it up for a PR yet. (By now its behind a few commits, but they should be irrelevant for your platform) You can try to use my fork, be aware that some API functions did get changed. Especially the begin() function no longer takes options, but now takes the baudrate. I just pushed some old commits that I missed that updated the examples and the readme with I hope all necessary info to get it working. CAN-FD frames should also work (even if an old comment states that it is not supported), some examples should document how to do it.
> 
> I used it successfully on the STM32G0B1KBUx, and briefly tested it on a few other ICs with the FDCAN peripheral. So it likely would work for you. Of course there might still be some sleeping bugs.

Do you plan on making PR about that, or should I try to include your additions to this library?

ShiPC123, 2025-08-20:
Sorry, I haven't been paying attention to the messages recently.
I will try to use the FDCAN library files. If you have time, I look forward to being able to adapt to it.